qmk/qmk_configurator #1338: [Feature Request] automatic generation of default keymaps

arrowmechanics, 2024-04-01:
### Description

Keeping keymaps in this repo seems to be counterproductive, since there are two sources of default keymaps: here and in the firmware  repo - ultimately this is duplicated code,  its just that  we keep it here in JSOHN, and 
 there in C.
 
Wouldnt it be convenient if the configurator just downloaded the default keymap automatically from  qmk_firmware source? Has this solution ever been discussed?

yanfali, 2024-04-01:
At the time it was very difficult to parse C, now we are data driven it is
probably time to revisit.

arrowmechanics, 2024-04-01:
Cool!
Adding an endpoint to the API for getting the default keymap parsed by CLI's `c2json` seems like the most obvious solution - it keeps the code-operations on the  API side, which already keeps itself updated with `qmk_firmware` code changes, right?.  

Do You think it's easily possible to add such endpoint to the API,  since the code is  not sourced?
We could take care of the configurator part then - for  fetching the keymap from the endpoint.