qmk/qmk_configurator #1340: [Bug]  Sorry, this doesn't appear to be a QMK keymap file.

jordanwegener, 2024-04-02:
### Describe the Bug

Seeing this error when trying to upload a keymap from a while ago, probably sometime in late 2022. I attempted to add the extra JSON fields that are present in a newly exported keymap file, but the layout appears incomplete with overlapping keys and such when imported.

Here's the file contents without my attempts to fix it:

```json
{
  "name": "Keychron Q10",
  "vendorProductId": 875823521,
  "macros": [
    "dssfsd",
    "{KC_LCTL,KC_LALT,KC_T}",
    "{KC_LGUI,KC_L} {KC_LCTL,KC_LALT,KC_L}",
    "{KC_LCTL,KC_A} {KC_LCTL,KC_C}",
    "garbage macro ",
    "{KC_LCTL,KC_LALT,KC_T}",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    "",
    ""
  ],
  "layers": [
    [
      "KC_MPLY",
      "KC_ESC",
      "KC_F1",
      "KC_F2",
      "KC_F3",
      "KC_F4",
      "KC_F5",
      "KC_F6",
      "KC_F7",
      "KC_F8",
      "KC_F9",
      "KC_F10",
      "KC_F11",
      "KC_F12",
      "KC_INS",
      "KC_PAUS",
      "KC_MYCM",
      "KC_GRV",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_MINS",
      "KC_EQL",
      "KC_BSPC",
      "KC_HOME",
      "KC_MENU",
      "KC_TAB",
      "KC_Q",
      "KC_W",
      "KC_E",
      "KC_R",
      "KC_T",
      "KC_Y",
      "KC_U",
      "KC_I",
      "KC_O",
      "KC_P",
      "KC_LBRC",
      "KC_RBRC",
      "KC_BSLS",
      "KC_PGUP",
      "KC_MS_BTN1",
      "KC_CAPS",
      "KC_A",
      "KC_S",
      "KC_D",
      "KC_F",
      "KC_G",
      "KC_NO",
      "KC_H",
      "KC_J",
      "KC_K",
      "KC_L",
      "KC_SCLN",
      "KC_QUOT",
      "KC_ENT",
      "KC_PGDN",
      "MACRO02",
      "KC_LSFT",
      "KC_NO",
      "KC_Z",
      "KC_X",
      "KC_C",
      "KC_V",
      "KC_B",
      "KC_DEL",
      "KC_N",
      "KC_M",
      "KC_COMM",
      "KC_DOT",
      "KC_SLSH",
      "KC_RSFT",
      "KC_UP",
      "MACRO01",
      "KC_LCTL",
      "MO(1)",
      "KC_LALT",
      "MO(3)",
      "KC_NO",
      "KC_NO",
      "KC_LGUI",
      "KC_SPC",
      "KC_RALT",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_LEFT",
      "KC_DOWN",
      "KC_RGHT"
    ],
    [
      "RGB_TOG",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_SLCK",
      "KC_PSCR",
      "KC_F17",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_F16",
      "RGB_TOG",
      "RGB_MOD",
      "RGB_VAI",
      "RGB_HUI",
      "RGB_SAI",
      "RGB_SPI",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_F15",
      "KC_TRNS",
      "RGB_RMOD",
      "RGB_VAD",
      "RGB_HUD",
      "RGB_SAD",
      "RGB_SPD",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_F14",
      "KC_TRNS",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "MAGIC_TOGGLE_NKRO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_F13",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "TO(19)",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "TO(16)",
      "KC_TRNS",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS"
    ],
    [
      "KC_MPLY",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_SPC",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS"
    ],
    [
      "KC_MPLY",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_SLCK",
      "KC_PSCR",
      "KC_F17",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_F16",
      "KC_MPRV",
      "KC_TRNS",
      "KC_TRNS",
      "KC_UP",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_PGUP",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_F15",
      "KC_MNXT",
      "KC_TRNS",
      "KC_LEFT",
      "KC_DOWN",
      "KC_RGHT",
      "KC_TRNS",
      "KC_NO",
      "KC_TRNS",
      "KC_HOME",
      "KC_PGDN",
      "KC_END",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_F14",
      "KC_TRNS",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_MRWD",
      "KC_MFFD",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_F13",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_UNDS",
      "KC_MINS",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS"
    ]
  ],
  "encoders": [
    [
      ["KC_VOLD", "KC_VOLU"],
      ["RGB_VAD", "RGB_VAI"],
      ["KC_VOLD", "KC_VOLU"],
      ["KC_MRWD", "KC_MFFD"]
    ]
  ]
}
```

noroadsleft, 2024-04-02:
I'm not 100% sure, but your JSON data looks more like a VIA keymap export.

phpbbireland, 2024-08-24:
Am I right in assuming the configuration can only be used with existing layout, that is, those already added to qmk git repo?

tzarc, 2024-08-24:
It can't be used with QMK, period. It's not a configurator export.

closed 2024-08-24 by tzarc