qmk/qmk_configurator #1361: [Bug] When i try to compile to my DraculaD - my oled not work

Artymonae, 2024-09-11:
### Describe the Bug

I try to use my keymap, and compile on site, but oled is turn off. Pls, help me, i need this keymap and default OLED on DraculaD
[draculad_draculad_layout_mine.json](https://github.com/user-attachments/files/16963337/draculad_draculad_layout_mine.json)

fauxpark, 2024-09-11:
You'll need to compile the firmware yourself. The Configurator has no support for configuring OLEDs, and they will not do anything if there is no default behaviour for them at the keyboard level.

closed 2024-09-11 by fauxpark

Artymonae, 2024-09-11:
> You'll need to compile the firmware yourself. The Configurator has no support for configuring OLEDs, and they will not do anything if there is no default behaviour for them at the keyboard level.

can you help me pls?

fauxpark, 2024-09-11:
https://docs.qmk.fm/newbs