qmk/qmk_configurator #1388: [Bug] Broken json when using layers higher than 7

rhusiev, 2025-01-05:
### Describe the Bug

When I use layers with numbers higher than 7, the json outputted is broken: sometimes it won't even download and sometimes it will download, but interpret the json as full of N/As later.

I attach an example of a working json with 6 layers. If you add an empty 8 layer with just, for example, a key 's' as the first one at https://config.qmk.fm/#/beekeeb/piantor_pro/LAYOUT_split_3x6_3, export the json and load the json, all the layers will become empty. If you do the same with layer 7, everything works, though

### Additional Context?

Working json:

```json
{
  "version": 1,
  "notes": "",
  "documentation": "\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n",
  "keyboard": "beekeeb/piantor_pro",
  "keymap": "beekeeb_piantor_pro_rad1an",
  "layout": "LAYOUT_split_3x6_3",
  "layers": [
    [
      "KC_TAB",
      "KC_F",
      "KC_P",
      "KC_D",
      "KC_L",
      "KC_X",
      "KC_SCLN",
      "KC_U",
      "KC_O",
      "KC_Y",
      "KC_B",
      "KC_Z",
      "KC_ESC",
      "KC_S",
      "KC_N",
      "KC_T",
      "KC_H",
      "KC_K",
      "KC_COMM",
      "KC_A",
      "KC_E",
      "KC_I",
      "KC_C",
      "KC_Q",
      "KC_LCTL",
      "KC_V",
      "KC_W",
      "KC_G",
      "KC_M",
      "KC_J",
      "KC_MINS",
      "KC_DOT",
      "KC_QUOT",
      "KC_EQL",
      "KC_SLSH",
      "KC_ENT",
      "KC_LALT",
      "LT(1,KC_LGUI)",
      "KC_R",
      "QK_REP",
      "LT(2,KC_SPC)",
      "OSL(3)"
    ],
    [
      "KC_TAB",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "LSFT(KC_SCLN)",
      "LSFT(KC_U)",
      "LSFT(KC_O)",
      "LSFT(KC_Y)",
      "LSFT(KC_B)",
      "LSFT(KC_Z)",
      "KC_ESC",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "LSFT(KC_COMM)",
      "LSFT(KC_A)",
      "LSFT(KC_E)",
      "LSFT(KC_I)",
      "LSFT(KC_C)",
      "LSFT(KC_Q)",
      "KC_LCTL",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "LSFT(KC_MINS)",
      "LSFT(KC_DOT)",
      "LSFT(KC_QUOT)",
      "LSFT(KC_EQL)",
      "LSFT(KC_SLSH)",
      "LSFT(KC_ENT)",
      "MO(5)",
      "KC_TRNS",
      "MO(4)",
      "LSFT(QK_REP)",
      "MO(6)",
      "MO(6)"
    ],
    [
      "RSFT(KC_TAB)",
      "RSFT(KC_F)",
      "RSFT(KC_P)",
      "RSFT(KC_D)",
      "RSFT(KC_L)",
      "RSFT(KC_X)",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "RSFT(KC_ESC)",
      "RSFT(KC_S)",
      "RSFT(KC_N)",
      "RSFT(KC_T)",
      "RSFT(KC_H)",
      "RSFT(KC_K)",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "RSFT(KC_LCTL)",
      "RSFT(KC_V)",
      "RSFT(KC_W)",
      "RSFT(KC_G)",
      "RSFT(KC_M)",
      "RSFT(KC_J)",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_LALT",
      "MO(6)",
      "RSFT(KC_R)",
      "KC_NO",
      "KC_TRNS",
      "MO(6)"
    ],
    [
      "KC_TAB",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_ESC",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_LCTL",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_LALT",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "MO(6)",
      "KC_TRNS"
    ],
    [
      "LGUI(KC_TAB)",
      "LGUI(KC_F)",
      "LGUI(KC_P)",
      "LGUI(KC_D)",
      "LGUI(KC_L)",
      "LGUI(KC_X)",
      "LGUI(KC_SCLN)",
      "LGUI(KC_U)",
      "LGUI(KC_O)",
      "LGUI(KC_Y)",
      "LGUI(KC_B)",
      "LGUI(KC_Z)",
      "LGUI(KC_ESC)",
      "LGUI(KC_S)",
      "LGUI(KC_N)",
      "LGUI(KC_T)",
      "LGUI(KC_H)",
      "LGUI(KC_K)",
      "LGUI(KC_COMM)",
      "LGUI(KC_A)",
      "LGUI(KC_E)",
      "LGUI(KC_I)",
      "LGUI(KC_C)",
      "LGUI(KC_Q)",
      "LGUI(KC_LCTL)",
      "LGUI(KC_V)",
      "LGUI(KC_W)",
      "LGUI(KC_G)",
      "LGUI(KC_M)",
      "LGUI(KC_J)",
      "LGUI(KC_MINS)",
      "LGUI(KC_DOT)",
      "LGUI(KC_QUOT)",
      "LGUI(KC_EQL)",
      "LGUI(KC_SLSH)",
      "LGUI(KC_ENT)",
      "LGUI(KC_R)",
      "KC_TRNS",
      "KC_TRNS",
      "QK_REP",
      "KC_RSFT",
      "KC_NO"
    ],
    [
      "LAG(KC_TAB)",
      "LAG(KC_F)",
      "LAG(KC_P)",
      "LAG(KC_D)",
      "LAG(KC_L)",
      "LAG(KC_X)",
      "LAG(KC_SCLN)",
      "LAG(KC_U)",
      "LAG(KC_O)",
      "LAG(KC_Y)",
      "LAG(KC_B)",
      "LAG(KC_Z)",
      "LAG(KC_ESC)",
      "LAG(KC_S)",
      "LAG(KC_N)",
      "LAG(KC_T)",
      "LAG(KC_H)",
      "LAG(KC_K)",
      "LAG(KC_COMM)",
      "LAG(KC_A)",
      "LAG(KC_E)",
      "LAG(KC_I)",
      "LAG(KC_C)",
      "LAG(KC_Q)",
      "LAG(KC_LCTL)",
      "LAG(KC_V)",
      "LAG(KC_W)",
      "LAG(KC_G)",
      "LAG(KC_M)",
      "LAG(KC_J)",
      "LAG(KC_MINS)",
      "LAG(KC_DOT)",
      "LAG(KC_QUOT)",
      "LAG(KC_EQL)",
      "LAG(KC_SLSH)",
      "LAG(KC_ENT)",
      "KC_TRNS",
      "KC_TRNS",
      "LAG(KC_NO)",
      "LAG(KC_NO)",
      "KC_RSFT",
      "KC_NO"
    ],
    [
      "QK_BOOT",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "DF(0)",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "LGUI(KC_SPC)",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS"
    ]
  ],
  "author": ""
}
```

yanfali, 2025-01-05:
What happens if you populate layer 7 and 8?

rhusiev, 2025-01-05:
Oh, thank you, it works! Do all previous layers have to be used before next ones? I think this behavior is unintuitive and destructive (I've lost my previous config this way)

yanfali, 2025-01-06:
I think it's a bug, it's a bit of a corner case. We don't allow empty
layers in a keymap I believe, so we should probably initialize intermediate
ones.

rhusiev, 2025-01-06:
Either initializing the layers automatically or erroring non-silently and telling that one can not export a keymap with empty intermediary layers would be good, I believe.