qmk/qmk_configurator #1394: [Bug] Keyboard Name Duplicated in Key Map Name

jack-faller, 2025-01-30:
### Describe the Bug

I was editing file using the configurator and uploaded/downloaded it a few times to make tweaks. I noticed that the name was set to "cantor_cantor_cantor_cantor_cantor_jackfaller" when it only started out as "jackfaller".

When uploading a keyboard, the key map in the configurator is set to the name specified in the JSON file, but when downloading a key map, they key map in the JSON has the keyboard name prepended to it. The result is that the keyboard name is duplicated every time a file is downloaded and uploaded.

E.g.
Start with keyboard = "keyboard" and name = "name", download key map, JSON = { "keymap": "keyboard_name" ... }, upload keymap, name = "keyboard_name", download keymap, JSON =  { "keymap": "keyboard_keyboard_name" ... }.

This makes it awkward to use as I always have to remove the duplicate keyboard name from the name field if I want to edit a JSON file multiple times.

yanfali, 2025-01-31:
Confirmed

yanfali, 2025-01-31:
Should be resolved shortly. Just a minor regression.

closed 2025-01-31 by yanfali