qmk/qmk_configurator #1397: [Bug] Incorrect mouse keycodes

eiis1000, 2025-03-16:
### Describe the Bug

The file `keymap_old.json` compiles just fine with `qmk compile -kb hillside/52 -km erez -e CONVERT_TO=promicro_rp2040`, but uploading it to QMK configurator, changing nothing, and downloading it yields `keymap_old_download.json`, which fails with the attached error. I've also attached a diff of the old and new files for convenience. For completion: `qmk --version` gives `1.1.6`.

[keymap_old.json](https://github.com/user-attachments/files/19274149/keymap_old.json)
[keymap_old_download.json](https://github.com/user-attachments/files/19274147/keymap_old_download.json)
[mouse_keycodes_broken_diff.txt](https://github.com/user-attachments/files/19274148/mouse_keycodes_broken_diff.txt)
[mouse_keycodes_broken_error.txt](https://github.com/user-attachments/files/19274146/mouse_keycodes_broken_error.txt)

yanfali, 2025-03-16:
@fauxpark were you making changes in this area?

tzarc, 2025-03-16:
The keycodes are correct, it's likely your local copy of QMK is out of date instead. Configurator has imported with the old names but exported with the new, up-to-date names.

These were added in QMK 0.26.0 to ensure (temporary) backwards-compatibility: [quantum_keycodes_legacy.h](https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum_keycodes_legacy.h)

eiis1000, 2025-03-17:
Thanks, I didn't realize that running setup/doctor/etc didn't auto-update.

closed 2025-03-17 by eiis1000